toleda/audio_CloverHDMI #1: Last cloverhdmi failed to download

grandosegood, 2016-02-12:
hello, with the latest version of your patch, i  does not appear to download properly.

using ga-x99-ud4, latest beta bios, 10.10.5 with x99 dmg 'debug applehda' from rampaged + all over files provided by his dmg. i get

Last login: Thu Feb 11 19:02:00 on ttys000
/Users/connoryosemite/Downloads/audio_cloverHDMI-110_v0.9.1.command ; exit;
connorymitesPro:~ connoryosemite$ /Users/connoryosemite/Downloads/audio_cloverHDMI-110_v0.9.1.command ; exit;
 
Agreement
The audio_cloverHDMI script is for personal use only. Do not distribute
the patch, any or all of the files or the resulting patched AppleHDA.kext
for any reason without permission. The audio_cloverHDMI script is
provided as is and without any kind of warranty.
 
File: audio_cloverHDMI-110.command_v0.9.1
EFI partition is mounted
Integrated Graphics is not installed/enabled
Nvidia discrete graphics card 10de13c0 found on 0/
usage: ioreg [-abfilrtx] [-c class] [-d depth] [-k key] [-n name] [-p plane] [-w width]
where options are:
	-a archive output
	-b show object name in bold
	-c list properties of objects with the given class
	-d limit tree to the given depth
	-f enable smart formatting
	-i show object inheritance
	-k list properties of objects with the given key
	-l list properties of all objects
	-n list properties of objects with the given name
	-p traverse registry over the given plane (IOService is default)
	-r show subtrees rooted by the given criteria
	-t show location of each subtree
	-w clip output to the given line width (0 is unlimited)
	-x show data and numbers as hexadecimal
Nvidia HDMI audio is not enabled
Install Nvidia HDMI audio ssdt (y/n): y
Download SSDT-HDMI-NVIDIA-0 ...
Password:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100     9  100     9    0     0     32      0 --:--:-- --:--:-- --:--:--    32
[/tmp/ssdt_hdmi-nvidia-0.zip]
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of /tmp/ssdt_hdmi-nvidia-0.zip or
        /tmp/ssdt_hdmi-nvidia-0.zip.zip, and cannot find /tmp/ssdt_hdmi-nvidia-0.zip.ZIP, period.
Error: Download failure, verify network.
No system files were changed
To save a Copy of this Terminal session: Terminal/Shell/Export Text As ...
logout

[Process completed]

bear in mind, my internet access is fine.

toleda, 2016-02-12:
README/cloverHDMI/6. Support/2. Intel/desktop series: 9, 8, 7, 6, 5
x99 is not supported at this time.

grandosegood, 2016-02-12:
toleda, thank you for the prompt reply. so, at this time, there is no way to use hdmi/displayport audio on an x99 system? regardless if nvidia or amd?

toleda, 2016-02-12:
x99/AMD/Nvidia/HDMI/DP audio works with dsdt edits/ssdt injection.  cloverHDMI/x99 support is planned for next release.  Note: any audio solution from this repo is not compatible with any rampaged ssdt.

grandosegood, 2016-02-12:
ok, i understand. i have a few questions about a temporary fix

1) do i need to use dsdt and ssdt together or can i use ssdt by itself.

2) i looked at some of your nvidia SSDTs, and noticed mine is GFX0 (GFX0@0)...if ssdt can be used by itself, should i rename one of yours or make my own? here is ioregg too just in case
[connoryosemite’s Mac Pro.zip](https://github.com/toleda/audio_CloverHDMI/files/127526/connoryosemite.s.Mac.Pro.zip)

toleda, 2016-02-12:
dsdt/ssdt: one or the other, both requires considerable care.

need ssdt/config.plist, otherwise, big guess

grandosegood, 2016-02-12:
ok, i was trying to use the ssdt provided by you along with the rampagedev ssdt. as previously stated, no cross mixing of guides, in his case, rampage dev and yours, can be supported. but, here is my ssdt and config, just in case you can think of a way
[github help.zip](https://github.com/toleda/audio_CloverHDMI/files/127736/github.help.zip)

toleda, 2016-02-12:
Current ssdt fix:
In: 
Device (_SB.PCI0.BR3A.GFX0)
Add:
                "@4,connector-type", 
                Buffer (0x04)
                {
                     0x00, 0x08, 0x00, 0x00                         
                }, 

                "@5,connector-type", 
                Buffer (0x04)
                {
                     0x00, 0x08, 0x00, 0x00                         
                }, 
edit:
            Return (Package (0x10)

Compile/Save/Install/Restart

grandosegood, 2016-02-12:
as they say in Metal Gear Solid..."You're Pretty Good!"

thank you very much, it has corrected hdmi audio, and i will check displayport later.

i have attached my edited SSDT, which uses the ga-x99-ud4 motherboard, 5820k and gtx 980.

closed 2016-02-12 by grandosegood