toleda/audio_CloverHDMI #2: There's anything for Sapphire HD 7750 Low Profile?

mths0x5f, 2016-03-20:
I have this card, OS X 10.11, but no HDMI sound. I have tried your script but no success. It seems setting Dashimaki do nothing, and card is recognized as HD 7XXX.

toleda, 2016-03-20:
Necessary, insufficient.
See:
1. https://github.com/toleda/audio_hdmi_guides/blob/master/%5BCase_Studies%5D_AMD-HD7xxx_HDMI_Audio_v2.pdf.zip
2. https://github.com/toleda/audio_hdmi_amd-nvidia/blob/master/ssdt_hdmi-amd/fb_injection/examples/ssdt_hdmi-amd-hd7750-PEG0.zip
3. https://github.com/toleda/audio_CloverHDMI/blob/master/config-hdmi_hd7750-100.plist.zip
4. http://www.tonymacx86.com/hdmi-audio/143760-audio-hdmi-audio-applehda-guide.html

ssdt edit required if the 7750 is not on PEG0.
find: PEG0, replace: ????

Old Post: http://www.tonymacx86.com/graphics/97540-no-dual-display-sapphire-7750-10-8-3-a.html#post596333

mths0x5f, 2016-03-21:
I'm sorry for the lack of information, but your help was needed. I'll try your guides, thank you!