toleda/audio_CloverHDMI #3: HDMI sound on HD520

chich0n, 2016-05-06:
Hello,

Is there any solution to make HDMI sound works on intel HD 520 ?(on a nuc6i3syh)

thank you

toleda, 2016-05-06:
Attach IOReg with working HD520.

chich0n, 2016-05-08:
Hello,

Thank you for your answer.
I send you my IOReg.

[chich0n_ioreg.zip](https://github.com/toleda/audio_CloverHDMI/files/254048/chich0n_ioreg.zip)

toleda, 2016-05-08:
Reply with cloverHDMI Terminal output.

chich0n, 2016-05-09:
My  terminal output : 
```
Agreement
The audio_cloverHDMI script is for personal use only. Do not distribute
the patch, any or all of the files or the resulting patched AppleHDA.kext
for any reason without permission. The audio_cloverHDMI script is
provided as is and without any kind of warranty.
 
File: audio_cloverHDMI-110.command_v0.20 - BETA
Release Mode
EFI partition is mounted
No HDMI audio codec(s) detected, continue (y/n): y
HDMI audio codec(s)
Device ID: 0x16190000 not supported
rm: /tmp/HDEF.txt: No such file or directory
AMD/Nvidia not found
No system files were changed
To save a Copy of this Terminal session: Terminal/Shell/Export Text As ...
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Opération terminée]

```

toleda, 2016-05-09:
My mistake.  
Not tested, make bootable backup, remove any existing HDMI audio files, edits, etc.
Download v0.21 and run.
Reply with Terminal output and IOReg.

chich0n, 2016-05-09:
thanks you for this new patch.
It doesn't work with 1.21.

```
Agreement
The audio_cloverHDMI script is for personal use only. Do not distribute
the patch, any or all of the files or the resulting patched AppleHDA.kext
for any reason without permission. The audio_cloverHDMI script is
provided as is and without any kind of warranty.
 
File: audio_cloverHDMI-110.command_v0.21 - BETA
Release Mode
EFI partition is mounted
No HDMI audio codec(s) detected, continue (y/n): y
HDMI audio codec(s)
HD515 HDMI audio is not enabled
Install SSDT-HDMI-HD515 HDMI audio ssdt (y/n): y
Download SSDT-HDMI-HD515 ...
Password:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 44727  100 44727    0     0  45331      0 --:--:-- --:--:-- --:--:-- 45316
Download SSDT-HDEF-HDAS-1.aml ...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 44816  100 44816    0     0  97133      0 --:--:-- --:--:-- --:--:-- 97004
/Volumes/EFI/EFI/CLOVER/ACPI/patched/SSDT-HDMI-HD515 installed
/Volumes/EFI/EFI/CLOVER/ACPI/patched/SSDT-HDEF-HDAS-1 installed
One display connected, proceeding
Confirm DP to HDMI connector edit on port 0x5 (y/n): y
Patch port 0x5
Download HD515 HDMI audio connector edits ...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1618  100  1618    0     0   4300      0 --:--:-- --:--:-- --:--:--  4303
/Users/florajussiau/Desktop/audio_cloverHDMI-110_v0.21.command: line 1293: patchconfig: command not found
HDMI audio edited /Volumes/EFI/EFI/CLOVER/config.plist installed
Error: discrete graphics card not found, unknown acpi PCI0 name
Names checked: 0 PEG0@1 PEG1@1 PEGP@1 P0P1@1 P0P2@1 P0P2@3 NPE3@2 NPE3@3 NPE7@3 pci-bridge@1 pci-bridge@3 BR3A@3
No system files were changed
To save a Copy of this Terminal session: Terminal/Shell/Export Text As ...
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Opération terminée]
```

But I think it's my fault. The first time i sent you an Ioreg, I use wrong injector (1912- Hd515 instead of 1916-HD520). So the patch can't work...

Thank you for you help. (and sorry for the mistake)
[ioreg_config_chich0n.zip](https://github.com/toleda/audio_CloverHDMI/files/255543/ioreg_config_chich0n.zip)

toleda, 2016-05-09:
Big problem, no IOReg/HD515 HDMI codec, not enabled.  Verify BIOS setting.
ssdt worked, script error fixed, download again and run.
Attach IOReg and config.plist

chich0n, 2016-05-09:
So I don't understand why my codec seems to be disable; It works fine under windows and HDMI audio is enable in BIOS.

chich0n, 2016-05-09:
I send you my IOReg and my config.plist :)

```
Agreement
The audio_cloverHDMI script is for personal use only. Do not distribute
the patch, any or all of the files or the resulting patched AppleHDA.kext
for any reason without permission. The audio_cloverHDMI script is
provided as is and without any kind of warranty.
 
File: audio_cloverHDMI-110.command_v0.21 - BETA
Release Mode
EFI partition is mounted
No HDMI audio codec(s) detected, continue (y/n): y
HDMI audio codec(s)
HD515 HDMI audio is not enabled
Install SSDT-HDMI-HD515 HDMI audio ssdt (y/n): y
Download SSDT-HDMI-HD515 ...
Password:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 44727  100 44727    0     0  39598      0  0:00:01  0:00:01 --:--:-- 39616
Download SSDT-HDEF-HDAS-1.aml ...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 44816  100 44816    0     0  76690      0 --:--:-- --:--:-- --:--:-- 76608
/Volumes/EFI/EFI/CLOVER/ACPI/patched/SSDT-HDMI-HD515 installed
/Volumes/EFI/EFI/CLOVER/ACPI/patched/SSDT-HDEF-HDAS-1 installed
One display connected, proceeding
Confirm DP to HDMI connector edit on port 0x5 (y/n): y
Patch port 0x5
Download HD515 HDMI audio connector edits ...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1618  100  1618    0     0   4017      0 --:--:-- --:--:-- --:--:--  4024
HDMI audio edited /Volumes/EFI/EFI/CLOVER/config.plist installed
Error: discrete graphics card not found, unknown acpi PCI0 name
Names checked: 0 PEG0@1 PEG1@1 PEGP@1 P0P1@1 P0P2@1 P0P2@3 NPE3@2 NPE3@3 NPE7@3 pci-bridge@1 pci-bridge@3 BR3A@3
No system files were changed
To save a Copy of this Terminal session: Terminal/Shell/Export Text As ...
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Opération terminée]
```
[chich0n.zip](https://github.com/toleda/audio_CloverHDMI/files/256004/chich0n.zip)

toleda, 2016-05-10:
Remove config.plist/ACPI/DSDT/Patches/:
1. 10.8+-Rename-GFX02IGPU
2. 10.11+-Rename-HECI2IMEI
3. Fix Skylake audio by renaming HDAS to HDEF
HD515 HDMI audio is configured properly.
DPCIManager shows ALC283 only, no HD515 HDMI audio codec; no HDMI audio,

chich0n, 2016-05-10:
I removed config's line : same result, no hdmi codec.
It's very strange

jungjinki, 2017-05-23:
Hello Toleda, 

Apology if my issue is out of topic here.  Nonetheless, this is most related to my issue I can find. 
I am not getting video to HDMI. My environment is El-Captian 10.11.6 running KabyLake / HD630
It would be appreciated if you can show me direction.  Please find the attached onionsjk-ioreg.zip for your reference. 

[onionsjk-ioreg.zip](https://github.com/toleda/audio_CloverHDMI/files/1021220/onionsjk-ioreg.zip)

toleda, 2017-05-23:
Hardware??
IOReg shows laptop display connected, no HDMI connected.
Connect HDMI display, restart, reply with new IOReg.

jungjinki, 2017-05-24:
Please find the attached for ioreg-with-HDMI-connected. 
I reboot with HDMI being connected. 

[onionsjk-ioreg-with-HDMI-connected.zip](https://github.com/toleda/audio_CloverHDMI/files/1024442/onionsjk-ioreg-with-HDMI-connected.zip)

toleda, 2017-05-24:
2nd request, what hardware?
IOReg does not show a HDMI device connected.
System Information/Hardware/Graphics/Displays will not show a HDMI device connected.

jungjinki, 2017-05-25:
I think a little bit of history is required at this time.                 

my notebook is Dell Inspiron 7567 KabyLake CPU running OS El-Capitan 10.11.6.
It has two graphic cards 1) Intel HD 630 and 2) nVidia 1050 Ti.

The first issue I faced was "7MB graphics".
Graphics is slow and no HDMI video.
Some applications I heavily use stop running and my suspicion was graphics.

I have tried nVidia Pascal driver but I couldn't make it.
But I managed to get "Intel Uknown 1536MB" using Intel HD option with help of
https://www.tonymacx86.com/threads/new-testing-intel-7th-generation-kaby-lake-cpus-200-series-motherboards-in-macos.219877/

Changed items are as below:
___- FackCPUID to 0x0506E3
__ - no acceleration  / I don't know how to set
__ - Intel HD 530 patch from MultiBeast
__ - Intel Graphics ID spoofed to 0x19168086 under /Devices/FakeID/IntelGFX
__ - FakePCIID.text and FakePCIID_Intel_HD_Graphics.ktext to /Library/Extensions
__ - AddProperty "Skylake Glitch Fix" to Devices/AddProperties/

Now graphics is fast and my application is running normal.
Only video to HDMI still not functioning.

Searching Dell Community site to find hardware info for HDMI, I find "Optimus" but
sorry I am not so sure it is right name or not.
http://en.community.dell.com/support-forums/laptop/f/3519/t/20007601

I hope this to be a help for trouble shooting.

I have a perfectly working notebook Dell Inspiron 7559 Skylake CPU
and can get its IOReg which will be useful for troubleshooting.
But it will be available on weekend only. Once done I will upload here.

toleda, 2017-05-25:
You have enabled Kaby Lake graphics, not Nvidia. macOS does not support Optimus or the device that switches graphics between Intel and Nvidia on your laptop.

jungjinki, 2017-05-26:
Please find the attached for IOReg of completely working notebook.
I saved it while I am having screen through HDMI. 
It is Dell 7559 SkyLake CPU with two graphics 1) Intel HD 530 2) nVidia GTX 960m running
ElCapitan-10.11.6. 
Its video to HDMI, recognition of graphics memory 1536MB and speed all are OK. 

[ioreg_Dell7559_ElCapitan_HDMI connected working.zip](https://github.com/toleda/audio_CloverHDMI/files/1033089/ioreg_Dell7559_ElCapitan_HDMI.connected.working.zip)