toleda/audio_CloverHDMI #4: Discrete graphics still not working

reyder, 2016-06-04:
I want to enable only HDMI audio for my AMD (I need inject Intel just for sleep).

Stage 1 choose no for Intel and after that I should see:
> Continue to AMD/Nvidia HDMI audio (y/n): y  (<<<<)

Strange thing is that this text it's not present in the file..so here we are:
`# initialize variable
choice1=n
choice2=n
gconnectoredit=n
gssdtinstall=n

if [ $igfxhdagfx = 2 ]; then  # ssdt working
    Echo "$gigfxname HDMI audio is enabled, connector edit may be required"
        while true
        do
        read -p "Verify $gigfxname HDMI audio connector/s (y/n): " choice1
        case "$choice1" in
            [yY]* ) gconnectoredit=y; break;;
            [nN]* ) gconnectoredit=n; break;;
            * ) echo "Try again...";;
        esac
        done
else
Echo "$gigfxname HDMI audio is not enabled"
        while true
        do
        read -p "Install $gigfxssdt HDMI audio ssdt (y/n): " choice2
        case "$choice2" in
            [yY]* ) gssdtinstall=y; break;;
            [nN]* ) gssdtinstall=n; break;;
            * ) echo "Try again...";;
        esac
        done
fi

gamdnvidia=n
if [[ $choice1 = n && $choice2 = n ]]; then
    if [ $gdgfx = 0 ]; then
        echo "AMD/Nvidia not foundnn"
        echo "No system files were changed"
        echo "To save a Copy of this Terminal session: Terminal/Shell/Export Text As ..."
        exit 1
    fi
fi
`
Choice 1 is default 0 and I don't want install Intel so choice 2 is also 0, result exit. Here is screenshot 

[http://imgur.com/a/ZaS7M](url)

toleda, 2016-06-04:
Post Terminal output.

reyder, 2016-06-04:
In screenshot @up.