toleda/audio_CloverHDMI #6: 10.12 - No HDMI audio codec(s) detected on NUC5i5RYH (HD6000)

lorenzolamasse, 2016-11-16:
Hi Toleda,

I have been trying without success to activate audio through HDMI, audio through Jack seems to work (ALC283 is used and I have used successfully he tool you provide).

On the ioreg output, apart from the ALC driver detailed the only other item is:
+-o IOHDACodecDevice@3,0  <class IOHDACodecDevice, id 0x1000002ed, !registered, !matched, active, busy 0, retain 5>
    {
    }
This "probably" refers to the HDMI output, but it is empty. Please advise... Maybe missing info in the DSDT ?
[ioreg.txt](https://github.com/toleda/audio_CloverHDMI/files/594513/ioreg.txt)

toleda, 2016-11-26:
Apologies for the delay. Use IORegistryExplorer.app/File/Save a Copy As...

lorenzolamasse, 2016-11-29:
Here it is. I have made some manual patching in the config.plist to try and activate it but with no success, so I attach it also.
[NUC.ioreg.zip](https://github.com/toleda/audio_CloverHDMI/files/619611/NUC.ioreg.zip)
[config.plist.zip](https://github.com/toleda/audio_CloverHDMI/files/619614/config.plist.zip)

toleda, 2016-11-29:
Clover HDMI audio does not work, cloverHDMI supports HD6000 HDMI audio
Back up what you have.  Remove the following properties from config.plist/
ACPI/DSDT/Patches/Item 4/change B0D3 to HDAU
ACPI/DSDT/Patches/Item 3/change GFX0 to IGPU
KernelAndKextPatches/KextsToPatch/Item 12/HDMI-audio, port 0204
KernelAndKextPatches/KextsToPatch/Item 11/HDMI-audio, port 0105
Save/Install
Restart
Run cloverHDMI
Restart,
Reply with new files

lorenzolamasse, 2016-11-30:
The thing is I get the "no HDMI audio codec(s) detected", should I proceed anyway ?

toleda, 2016-11-30:
No, the HDMI audio codec is not support natively.

Try this patch:
config.plist/KernelAndKextPatches/KextsToPatch/
Comment: 10.11+-HD6000_HDMI_Audio
Find:	3d0c160000
Name:	AppleHDAController
Replace:	3d0a160000

Save/install/restart
Reply with new IOReg

lorenzolamasse, 2016-11-30:
No change in behaviour, here is the ioreg (and config.plist in case...)
[config.plist.zip](https://github.com/toleda/audio_CloverHDMI/files/622263/config.plist.zip)
[NUC.ioreg.zip](https://github.com/toleda/audio_CloverHDMI/files/622266/NUC.ioreg.zip)

toleda, 2016-11-30:
Graphics did not load. Set config.plist/Graphics/Inject/Intel/YES.
Reply with new files and DCPIManager/Status screenshot.

lorenzolamasse, 2016-12-01:
As required...
[image]

toleda, 2016-12-02:
Clover/Intel graphics detection did not work.
Install: https://github.com/toleda/audio_hdmi_9series/blob/master/ssdt_hdmi-hd6000%2B/ssdt_hdmi-hd6000.zip
Reply with new files.

lorenzolamasse, 2016-12-02:
[config.plist.zip](https://github.com/toleda/audio_CloverHDMI/files/627715/config.plist.zip)
[image]
[NUC.ioreg.zip](https://github.com/toleda/audio_CloverHDMI/files/627716/NUC.ioreg.zip)
Here are the new files... No change whatsoever I would say... Is there a way to make sure the SSDT loads properly ? (no "sorted order" in config.plist for example).

toleda, 2016-12-02:
Attach clover boot log.

lorenzolamasse, 2016-12-03:
[debug.log.zip](https://github.com/toleda/audio_CloverHDMI/files/628798/debug.log.zip)
Here it is...

toleda, 2016-12-03:
Compress EFI/CLOVER/ACPI/patched and attach
Compress MaciASL/File/Export Tableset.... and attach

lorenzolamasse, 2016-12-05:
there you go. thank you for your time...

[patched.zip](https://github.com/toleda/audio_CloverHDMI/files/630547/patched.zip)
[NUC.acpi.zip](https://github.com/toleda/audio_CloverHDMI/files/630552/NUC.acpi.zip)

toleda, 2016-12-05:
Haswell HDMI audio ssdts installed.
Remove patched/1. SSDT-IGPU.aml and 2. SSDT-HDAU.aml
Reply with config.plist, IOReg and Clover boot log.

lorenzolamasse, 2016-12-06:
[config.plist.zip](https://github.com/toleda/audio_CloverHDMI/files/633461/config.plist.zip)
[debug.log.zip](https://github.com/toleda/audio_CloverHDMI/files/633464/debug.log.zip)
[NUC.ioreg.zip](https://github.com/toleda/audio_CloverHDMI/files/633465/NUC.ioreg.zip)
Here it is...

toleda, 2016-12-06:
Remove patched/SSDT-HDMI-HD6000.aml
Install attached SSDT-HDMI-HD6000.aml
Reply with new files
[ssdt_hdmi-hd6000+_v2.zip](https://github.com/toleda/audio_CloverHDMI/files/634599/ssdt_hdmi-hd6000._v2.zip)

lorenzolamasse, 2016-12-07:
[config.plist.zip](https://github.com/toleda/audio_CloverHDMI/files/636359/config.plist.zip)
[debug.log.zip](https://github.com/toleda/audio_CloverHDMI/files/636362/debug.log.zip)
[NUC.ioreg.zip](https://github.com/toleda/audio_CloverHDMI/files/636366/NUC.ioreg.zip)
From a sound perspective I see no change, but I have now lost Full HD, stuck at 1280x1024 :p

toleda, 2016-12-07:
My mistake, try 
[ssdt_hdmi-hd6000+_v3.zip](https://github.com/toleda/audio_CloverHDMI/files/636861/ssdt_hdmi-hd6000._v3.zip)

lorenzolamasse, 2016-12-07:
[NUC.ioreg.zip](https://github.com/toleda/audio_CloverHDMI/files/636886/NUC.ioreg.zip)
[debug.log.zip](https://github.com/toleda/audio_CloverHDMI/files/636887/debug.log.zip)
[config.plist.zip](https://github.com/toleda/audio_CloverHDMI/files/636888/config.plist.zip)
No worries, there you go...

toleda, 2016-12-07:
Try [ssdt_hdmi-hd6000+_v4.zip](https://github.com/toleda/audio_CloverHDMI/files/637995/ssdt_hdmi-hd6000._v4.zip)

lorenzolamasse, 2016-12-08:
[config.plist.zip](https://github.com/toleda/audio_CloverHDMI/files/638586/config.plist.zip)
[debug.log.zip](https://github.com/toleda/audio_CloverHDMI/files/638589/debug.log.zip)
[NUC.ioreg.zip](https://github.com/toleda/audio_CloverHDMI/files/638592/NUC.ioreg.zip)
Second Audio Device has totally disappeared now, I only have the Relatek in DPCIManager...

toleda, 2016-12-08:
Mistake. Remove config, plist/KernelAndKextPatches/KextsToPatch//10.11+-HD6000_HDMI_Audio
Try [ssdt_hdmi-hd6000+.zip](https://github.com/toleda/audio_CloverHDMI/files/640415/ssdt_hdmi-hd6000.zip)

lorenzolamasse, 2016-12-08:
[config.plist.zip](https://github.com/toleda/audio_CloverHDMI/files/640494/config.plist.zip)
[debug.log.zip](https://github.com/toleda/audio_CloverHDMI/files/640497/debug.log.zip)
[NUC.ioreg.zip](https://github.com/toleda/audio_CloverHDMI/files/640498/NUC.ioreg.zip)
Alas, it does not get any better :(

toleda, 2016-12-08:
HD6200 HDMI audio is working with the same codec. Do not know why NUC/HD6000 HDMI audio is not working.  
Saw the problem with Asus UEFI/P67/HD3000; only solution is a BIOS downgrade.
No further suggestions.

lorenzolamasse, 2016-12-08:
thx, I will check if the platform accepts such a downgrade...

toleda, 2016-12-18:
Try https://github.com/RehabMan/Intel-NUC-DSDT-Patch (NUC5)