toleda/audio_CloverHDMI #7: audio_cloverALC vs audio_cloverHDMI

RhanCandia, 2016-11-26:
On my build, I used audio_cloverALC to fix audio. HD audio works but not HDMI audio. Can I used this script on top of audio_cloverALC?

toleda, 2016-11-26:
Yes, as designed.

RhanCandia, 2016-11-26:
Hi,

Thanks for the response. I will give it a try. I jad to asl because I was reading the markup and I didn't see cloverALC mentioned.

Hope this fixes my HD 530 HDMI audio.

toleda, 2016-11-26:
If installed, remove config.plist/ACPI/DSDT/Patches/Item ?/Rename HDAS to HDEF; disables Skylake HDMI audio ssdt. 
Install https://github.com/toleda/audio_ALCInjection/blob/master/ssdt_hdef/ssdt_hdef-1-100-hdas.zip

RhanCandia, 2016-11-27:
Can't find this part "config.plist/ACPI/DSDT/Patches/Item ?/Rename HDAS to HDEF" on my config.plist

toleda, 2016-11-27:
No.  How did you set dsdt/HDAS to HDEF?

RhanCandia, 2016-11-27:
Here goes.

Fresh install of Sierra. After setting up clover and other clover configs and kext. I ran audio_cloverALC. after that, I rebooted and HD Audio works. I didn't do anything else from there. But when I pluged into HDMI I noticed it doesn't have audio. So I looked it up and found this. I never messed with DSDTs. Other than generating an SSDT fpr my CPU.

RhanCandia, 2016-11-28:
@toleda here's my config by the way.

http://pastebin.com/z3w7yXR8

toleda, 2016-11-28:
No evidence you ran cloverHDMI, why not?
Instead config.plist shows intelHDMI set, does not work.
Remove config.plist (ignore line numbers/Pastebin)
	1	            <key>Fixes</key>
	2	            <dict>
	3	                <key>AddHDMI_8000000</key>
	4	                <true/>
	5	                <key>FixHDA_8000</key>
	6	                <true/>
	7	    <key>Devices</key>
	8	    <dict>
	9	        <key>AddProperties</key>
	10	        <array>
	11	            <dict>
	12	                <key>Device</key>
	13	                <string>IntelGFX</string>
	14	                <key>Key</key>
	15	                <string>AAPL,GfxYTile</string>
	16	                <key>Value</key>
	17	                <data>
	18	                AQAAAA==
	19	                </data>
	20	            </dict>
	21	        </array>
Run cloverHDMI
For problems, README/Problem Reporting, attach all requested files.

RhanCandia, 2016-11-29:
@toleda Oops. My mistake. The config I pasted was the one BEFORE I ran cloverHDMI. I will do again later when I get home.

toleda, 2016-11-29:
No.
cloverHDMI installs ssdts and patches config.plist as required.

RhanCandia, 2016-11-29:
@toleda alright. Then having nothing under ACPI / DSDT / paches on Clover Configurator is normal then.

Maybe I can briefly explain what happens when I run cloverHDMI script.

1. I ran cloverHDMI.
2. It says Intel HD 530 HDMI audio is not enabled. Would I like to enable? I say "Y" for yes. Then I say no for DP to HDMI, and ATI/NVIDIA hdmi. Because I don't use DP and a video card. So that Y, N and N.
3. I can see that it installs 2 SSDTs on EFI/CLOVER/ACPI/patched
4. But It doesn't change anything on my clover config. Why is that?

toleda, 2016-11-29:
Native framebuffer only supports DP
If there is no DP connected, the answer for DP to HDMI must be Y. Y edits config.plist.
cloverHDMI replies for your system are Y, Y, N.

oscarssama, 2017-12-03:
@toleda almost same problem as @RhanCandia 

But I went Y/Y/N for the Intel HD 530, seems to copy the hdmi file to EFI/CLOVER/ACPI/patched but after restart there is no output for hdmi