toleda/audio_CloverHDMI #8: Scripts doesn't work, if user has moved his home folder to another drive

mrPsycho, 2016-12-08:
Cause not everybody places home directory in /Users/$(whoami) .

toleda, 2016-12-08:
cloverALC does not touch the home directory; what is the problem?

mrPsycho, 2016-12-09:
check audio_cloverHDMI-120_v1.0d.command:

`gCloverDirectory=/Volumes/$gStartupDisk/EFI/CLOVER`
works incorrect on my system :( 
i had to write exact `gCloverDirectory=/Volumes/EFI/EFI/CLOVER`

and:
`gDesktopDirectory=/Volumes/$(whoami)/Desktop`

`1 )
    if [ -d "$gDesktopDirectory/config-basic.plist" ]; then
       echo "Desktop/config-basic.plist copied missing"
       exit 1
    fi

    sudo cp -R "$gDesktopDirectory/config-basic.plist" /tmp/config.plist
    echo "Desktop/config-basic.plist copied to /tmp/config.plist"
   ;;

2 )
    echo "gHDAversioninstalled = $gHDAversioninstalled"
    if [ -d "$gDesktopDirectory/config-basic.plist" ]; then
       echo "Desktop/config-basic.plist copied missing"
       exit 1
    fi

    sudo cp -R "$gDesktopDirectory/config-basic.plist" /tmp/config.plist
    echo "Desktop/config-basic.plist copied to /tmp/config.plist"
    ;;

* )
    echo "gDebug = $gDebug, fix"
    exit 1
    ;;`

toleda, 2016-12-10:
What script do you use to mount the EFI partition?
 Attach terminal output with the error.

mrPsycho, 2016-12-15:
sorry for delay,
i'm using Clover Configurator

toleda, 2016-12-15:
Terminal output?

mrPsycho, 2016-12-21:
ok. just updateded macos to the newest version.
so:

    sy:~ mrpsycho$ sudo /Volumes/data/Users/mrpsycho/Downloads/audio_cloverHDMI-120_v1.0d.command
    Password:

    Agreement
    The audio_cloverHDMI script is for personal use only. Do not distribute
    the patch, any or all of the files for any reason without permission.
    The audio_cloverHDMI script is provided as is and without any warranty.

    File: audio_cloverHDMI-120.command_v1.0d
    Release Mode
    EFI partition is mounted
    HDMI audio codec(s)
    Nvidia:   0x10de0060
    Integrated Graphics is not installed/enabled
    Install AMD/Nvidia HDMI audio (y/n): y
    Nvidia discrete graphics card 10de1381 found on PEG0@1/display@0
    Nvidia HDMI audio is not enabled
    Install Nvidia HDMI audio ssdt (y/n): y
    Download SSDT-HDMI-NVIDIA-PEG0 ...
     % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
    100 44888  100 44888    0     0   131k      0 --:--:-- --:--:-- --:--:--  132k
/Volumes/EFI/EFI/CLOVER/ACPI/patched/SSDT-HDMI-NVIDIA-PEG0.aml installed

    Install finished, restart required.
    To save a Copy of this Terminal session: Terminal/Shell/Export Text As ...

mrPsycho, 2016-12-21:
so it seems to be ok, but no sound from my nvidia card thru hdmi :(

toleda, 2017-01-24:
Apologies for the delay, files?