toleda/audio_CloverHDMI #11: HDMI active for a NUC6I3SYH

MisterM74, 2017-03-11:
Hello
How to activate, no, is it activated on a NUC6I3SYH HDMI Audio?
IF yes, how, thank you

Mz

toleda, 2017-03-11:
Try cloverHDMI. If successful or not, attach Problem Reporting Files.

MisterM74, 2017-03-12:
Hello
Here is a correct answer.
So, having found the console to see the logs, it log file I have to check to give you and posted what works?
Thank you
Mz

toleda, 2017-03-12:
See README/Problem Reporting.  Attach requested files.

MisterM74, 2017-03-12:
Hello

With this audio_codecdetect.command, result

`HDMI audio codec(s)

Onboard audio codec
Realtek: 0x10ec0283
Name: Realtek ALC283
Audio ID: Not supported
Current Audio ID: 2 (valid: 1, 2, or 3)
FIX, try:
1. https://github.com/toleda/audio_ALCInjection/tree/master/ssdt_hdef/
ssdt_hdef-1-with_ioreg/hdef.zip (select View Raw)
2. CLOVER/config.plist/Devices/Audio/Inject/1

No system files were changed
To save a Copy of this Terminal session: Terminal/Shell/Export Text As ...
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...none found.`

Mz

toleda, 2017-03-12:
Use Audio ID: 1 not 2.

MisterM74, 2017-03-13:
Hello
This:

> Use Audio ID: 1 not 2.

Or I change it in the config.plist file?
Thank you
Mz

toleda, 2017-03-13:
Edit config.plist/Devices/Audio/Inject/1

MisterM74, 2017-03-13:
Hello

I suspected it was the right thing to change, unfortunately, it changes nothing after restart.

It's stupid, because the rest works, there is that that does not work ...

Question, your plugin, works for the 2, UEFI, Legacy.

Mz

toleda, 2017-03-13:
Yes. run cloverHDMI
See README/Problem Reporting. Attach requested files.