toleda/audio_CloverHDMI #12: where is ssdt

RoJoHub, 2017-04-27:
```
- Installs HDMI audio ssdt(s) and edited config.plist
  - Intel/AMD/Nvidia: HDMI audio ssdt (EFI/CLOVER/ACPI/patched/)
```
where is ssdt？sorry,My english is not good

toleda, 2017-04-27:
EFI/CLOVER/ACPI/patched/

RoJoHub, 2017-04-27:
@toleda 
There is only one file in the folder.And the file name is `SSDT-SB-OC.aml`
I guess the file is install when i open `MultiBeast`

toleda, 2017-04-27:
cloverHDMI installs the required HDMI audio ssdt.

RoJoHub, 2017-04-27:
Also i have another question.
How to set /Debug
```
##Test Drive
- Set audio_cloverHDMI-xxx.command/gDebug=1
```

Yesterday I update 10.12.4. Afterwards i ran `audio_cloverHDMI-120_v1.0d.command`.Rest.
My hackintosh is still process.
So i reinstall  hackintosh.
Now I have come to this step.
I remember my hackintosh 10.12.2 run `audio_cloverHDMI-120_v1.0d.command` is Ok.

closed 2017-04-28 by RoJoHub