toleda/audio_CloverHDMI #13: Question:10.12.4

RoJoHub, 2017-04-28:
@toleda.
Also i have another question.
How to Set audio_cloverHDMI-xxx.command/gDebug=1
```
Test Drive
- Set audio_cloverHDMI-xxx.command/gDebug=1
```
The day before yesterday I update 10.12.4. Afterwards i ran `audio_cloverHDMI-120_v1.0d.command` once again. 
My hackintosh is still process after restart machine.
So i reinstall hackintosh.
Now I have come to this step.
I remember my hackintosh 10.12.2 run `audio_cloverHDMI-120_v1.0d.command` is Ok.
Thanks for your help

RoJoHub, 2017-04-28:
There is an option in the menu in MultiBeast.
[image]
```
 - I.cloverHDMI (script)
    - detects and installs appropriate ssdt(s)
```
So i think I finished this step.
After all of run.
`Using Intel HD530 with Dual Monitors`is not working
But hdmi is effective.

toleda, 2017-04-29:
For debug=1
1. open the script with Xcode or a text editor
2. find gDebug=0
3. set gDebug=1
4. save

For any install, run cloverHDMI once. Software Update has no effect on ACPI edits.

The HD3000/HD4000 selection sets Audio ID = 3, does not enable HDMI audio.  Does not apply to HD530, description error.

See README/Problem Reporting, attach requested files.

closed 2021-03-23 by RoJoHub