toleda/audio_CloverHDMI #16: HD630 support

malhal, 2017-08-18:
Hi I have a HD630 via i7-7700 and H270 chipset and happy to help with testing HDMI audio. I saw the Kabylake commit with config-hdmi_hd6x0-120.plist.zip however I didn't understand the instructions in the readme on how to use it. Step 1 mentions running a command which errors with Device ID: 0x12590000 not supported, and Step 2 didn't make sense to me.

I thought about doing a diff between hd6x0 and my config and then incorporating the differences but thought I would ask you first in case you have yet to commit the change to the command script.

Thanks

toleda, 2017-08-19:
UseIntelHDMI may add injections, however, not sufficient for working HDMI audio. Often, arbitrary settings override/disable other injections; use with caution.  cloverHDMI-130 supports Kaby Lake/HD6x0, without arbitrary injection.

malhal, 2017-08-19:
Sorry I'm a bit confused by that, I added 2 Arbitrary settings from the config-hdmi_hd6x0-126.plist inside config-hdmi_hd6x0-120.plist.zip. They were the 10.8+-Intel-IGPU-HDMI-HDA and 10.12.6-Intel-HDA-Desktop ones. Are you saying I don't need those?

toleda, 2017-08-19:
My mistake, fixed.

malhal, 2017-08-22:
Removed all my manual edits, which included the renames HDAS (although I had that previous to attempting to fix HDMI audio but thought would check if this command adds it), GFX, HECI. Those 2 Arbitrarys I mentioned above, and 10.12.6-KBL-5912000-Port_0x5-DP2HDM.

The command didn't output what changes it made but I did a diff and saw that it only made this one change: 

```
		<key>KextsToPatch</key>
		<array>
			<dict>
				<key>Comment</key>
				<string>t2-10.12.6-KBL-5912000-Port_0x6-DP2HDM</string>
				<key>Find</key>
				<data>AgQKAAAEAACHAQAA</data>
				<key>Name</key>
				<string>AppleIntelKBLGraphicsFramebuffer</string>
				<key>Replace</key>
				<data>AgQKAAAIAACHAQAA</data>
			</dict>
```

Is it correct that it should only make that one change? Also I noticed previously it was using port 5 and now is port 6 is that ok too? I saw some curl output, is it downloading the config-hdmi_hd6x0-120.plist.zip from the repo and then replacing values? That would explain why the number is different from what I took manually out of it the first time.

toleda, 2017-08-22:
Kaby Lake/HD630 support only available in cloverALC-130.  Sky Lake desktop framebuffer required a second patch to fix a framebuffer error, Kaby Lake desktop framebuffer is correct with one patch. Attach IOReg for a valid reply to the port question.

malhal, 2017-08-22:
Apologies, it might sound strange but I didn't even realise that cloverALC was a pre-requisite for cloverHDMI. I was using a different patch for my audio but I will start over with exclusively your patches and see how that goes.

toleda, 2017-08-22:
Apologies, major typo. Meant cloverHDMI-130. cloverALC is not a prerequisite for HDMI audio.

malhal, 2017-08-22:
No worries about that. So I cleared out any previous audio patches, including some stuff I found in the config that wasn't even on the Configurator UI (e.g. IOKitPersonalities with about 20 entries mentioning toledaALC maybe came from Unibeast). Applied cloverALC after reboot audio worked. Then applied cloverHDMI but after reboot no additional HDMI audio appeared. Seems I'm missing something else which I did have when I did it manually. I've attached all relevant files, thanks.

[malhal.zip](https://github.com/toleda/audio_CloverHDMI/files/1242945/malhal.zip)

malhal, 2017-08-22:
Didn't even know I needed audio_ALCInjection but I applied that and regular audio is working. HDMI audio is still not an option though.

toleda, 2017-08-22:
HDMI audio, 2 issues:
1. Remove config.plist/ACPI/DSDT/Patches/Item 0 (Rename HDAS to HDEF)
2. Port 0x5 patch is correct. Fix, download v0.3 from repo (same link as above)
Restart

malhal, 2017-08-22:
Syntax error, extra bracket on line 1053.

But with the bracket fixed and the steps followed sorry to report that all audio outputs have gone from sound prefs.
[malhal config.plist.zip](https://github.com/toleda/audio_CloverHDMI/files/1243181/malhal.config.plist.zip)

toleda, 2017-08-22:
Apologies, I am traveling and do not haver access to my development machine.
config.plist shows the correct framebuffer edit, useless to figure out what changed.
Attach cloverHDMI output and IOReg.

malhal, 2017-08-22:
No worries, here is a zip with the 2 files requested.

[malhal ioreg cloverHDMI output.zip](https://github.com/toleda/audio_CloverHDMI/files/1243221/malhal.ioreg.cloverHDMI.output.zip)

malhal, 2017-08-22:
I thought we would have needed the HDAS to HDEF and the UseIntelHDMI but interested to hear what you think. No rush at all enjoy your travels!

toleda, 2017-08-22:
1220: realtelALC.kext is missing.
Verify realtekALC.kext is installed in EFI/CLOVER/kexts/10.12 (or Other) or /L/E
HDMI audio: What display is connected to what motherboard connector?

cloverHDMI installs EFI/CLOVER/ACPI/patched/SSDT-HDEF-HDAS-1 which does what the Clover is supposed to do but doesn't.

malhal, 2017-08-22:
The aml isn't installed to that dir, will try to debug why, not great at bash tho. I have the aml -1 and -2 files from earlier tho so at least can try this out. Sorry yes I had deleted realtekALC because I thought AppleHDA has support for the 1220 now.

toleda, 2017-08-23:
Use aml-1 only
Removed 1220 support, run cloverALC to restore realtekALC.kext

malhal, 2017-08-24:
Hi even with realtekALC added back I still haven't been able to get it to work so I reverted back to my old technique for the moment. I'll try this again soon though and post the output from each step again.

malhal, 2017-08-24:
Ok I have figured it out. After clearing everything out here is why it goes wrong:

1. Run cloverALC, errors no HDEF
2. Run cloverHDMI, errors no HDEF
3. Manually copy aml-1 reboot
4. Run cloverALC, success, reboot and audio works
5. Run cloverHDMI, success, reboot and audio is gone!!!
6. Manually delete aml-2, reboot and both audio and hdmi audio is working. (I would have never figured this out without your comment because at this stage there are too many variables).

I also had ran ALCInjection audio_200_Series_hda-124_v1.0.command early on and I think maybe it didn't work without the HDEF either.

So I think to fix it cloverALC or ALCInjection needs to install the aml-1. And cloverHDMI should not install aml-2.

If you'd like me to run through it again and provide the terminal output from each step I'll happily do that.

toleda, 2017-08-25:
HD630 HDMI audio requires HDEF
ALCInjection audio_200_Series_hda-124_v1.0.command is one of 4 methods to enable Kaby Lake onboard audio, nothing to do with HD630 HDMI audio.
Without cloverHDMI output, no idea how SSDT-HDEF-HDAS-2.aml was installed; not applicable to your system.

malhal, 2017-08-25:
Here is requested output:
[cloverHDMI output.txt](https://github.com/toleda/audio_CloverHDMI/files/1250578/cloverHDMI.output.txt)

toleda, 2017-08-25:
config.plist/Devices/Audio/Inject/3
Why is Audio ID: 3?

malhal, 2017-08-25:
Almost certain cloverALC asked "Use Audio ID: 3?" Maybe at 3. I accidentally used SSDT-HDEF-HDAS-3.aml Would that have done it?

toleda, 2017-08-25:
Yes, 1220 does not support Audio ID: 3.
cloverALC does not allow Audio ID: 3 for 1220

malhal, 2017-08-25:
Yep I had SSDT-HDEF-HDAS-3.aml in patches by mistake but as you can see it did suggest and allow 3 for ALC1220 maybe it could not do that?
```
Confirm Realtek ALC1220 (y/n): y
Clover Audio ID Injection (y/n): y
Use Audio ID: 3 (y/n): y
```
[cloverALC audio3.txt](https://github.com/toleda/audio_CloverHDMI/files/1251411/cloverALC.audio3.txt)

toleda, 2017-08-26:
Stand corrected, checks when response is n; fix in process. Every previous attached config.plist had Audio ID: 1.

johnshine, 2017-08-29:
The audio_cloverHDMI-130.command generate 3 files:
HD630-config.plist
HD630-SSDT-HDMI-HD630.aml
HD630-SSDT-HDEF-HDAS-3.aml

I know what to with the plist file, but how to install other 2 aml file? just place in COVER->ACPI->patched folder?

toleda, 2017-08-29:
Rename HD630-config.plist to config.plist, otherwise Clover ignores the file.
script does not add HD630 to the .aml file name, remove the HD630 ... files.
Edit onfig.plist/Devices/Audio/Inject/1, save
run cloverHDMI

johnshine, 2017-09-01:
Follow your guid, but still not work. Here is my ioreg and config.plist, hope for your help. Thanks!
[johnshine.zip](https://github.com/toleda/audio_CloverHDMI/files/1270817/johnshine.zip)

toleda, 2017-09-01:
IOReg is corrupt, use v2.1 and verify.

johnshine, 2017-09-02:
Remove Item 3 (Change GFX0 2 IGPU), computer cannot startup. So I only Item 2 (Rename HDAS to HDEF) (by disabled), following is new files:

[johnshine.zip](https://github.com/toleda/audio_CloverHDMI/files/1271882/johnshine.zip)

toleda, 2017-09-02:
What happened to EFI/CLOVER/ACPI/patched/
1. SSDT-HDEF-HDAS-1.aml ?
2. SSDT-HDMI-HD630.aml ?

Attach cloverHDMI terminal output and Clover boot log.

johnshine, 2017-09-02:
No any aml in patched folder there. Here is my Clover folder at EFI partition & boot log

[Clover.zip](https://github.com/toleda/audio_CloverHDMI/files/1271939/Clover.zip)

johnshine, 2017-09-02:
ioreg export by explorer v2.1 with & without Item 2 (Rename HDAS to HDEF) patch
[ioreg files.zip](https://github.com/toleda/audio_CloverHDMI/files/1271980/ioreg.files.zip)

toleda, 2017-09-02:
Clover.zip shows no ssts; User error, fail.
Steps:
1. Mount EFI partition
2. REMOVE Rename HDAS to HDEF and Change GFX0 2 IGPU patches and save
2. RUN cloverHDMI
3. Confirm SSDT-HDEF-HDAS-1.aml  and SSDT-HDMI-HD630.aml installed 
4. Restart
5. Reply with EFI/CLOVER folder, clover boot log and IOReg.

johnshine, 2017-09-05:
Where can I get SSDT-HDEF-HDAS-1.aml and SSDT-HDMI-HD630.aml? RUN audio_cloverHDMI-130.command only get HD630-SSDT-HDEF-HDAS-3.aml and HD630-SSDT-HDMI-HD630.aml. I not quite understand.

toleda, 2017-09-05:
cloverHDMI does not install HD630-SSDT-HDEF-HDAS-3.am or HD630-SSDT-HDMI-HD630.aml, remove both
The HDAS-3 bug is fixed in the latest  cloverHDMI, download latest.
Follow Steps 1-6 above.

johnshine, 2017-09-06:
latest cloverHDMI has report errors, the file it downloaded as 404 not found in github. Follow is terminal output

[terminal output.txt](https://github.com/toleda/audio_CloverHDMI/files/1281151/terminal.output.txt)

toleda, 2017-09-06:
Is config.plist/Devices/Audio/Inject/4? If yes, why?
If no, attach IOReg, config.plist and Clover boot log.

johnshine, 2017-09-07:
Yes, because 4 is for desktop according doc

john-shine, 2017-09-07:
It Works now, great thanks to you