toleda/audio_CloverHDMI #17: XFX R9 280X DisplayPort audio

malhal, 2017-08-31:
Hi I was unable to get DisplayPort audio working on this card. From the readme it is a ambiguous if DP audio is supported. Into says "Clover HDMI audio enables HDMI, DP and DVI audio", step 4 says HDMI only. I ran the command and said yes to AMD HDMI and it downloaded the PEG0 aml.  If it should work I can post files to get it working.

I found a mysterious file on the web called HDMIAudio.kext and it was successful however I would prefer to do it with your SSDTs if possible, thanks.

toleda, 2017-08-31:
See README/Problem Reporting, attach requested files.
AMD does not support DVI audio; HDMI and DP audio supported.

malhal, 2017-08-31:
audio_cloverHDMI-130.command.zip contains audio_cloverALC-130_v0.2.command by mistake

toleda, 2017-08-31:
Apologies, fixed. Download latest.

malhal, 2017-08-31:
No worries, ok output from new audio_cloverHDMI-130_v0.4.command attached, config and IOReg after reboot. 

[Archive.zip](https://github.com/toleda/audio_CloverHDMI/files/1268310/Archive.zip)

toleda, 2017-08-31:
Motherboard? 
MaciASL/File/Exract Tableset ...
Attach .acpi file
Problem is dsdt, more info: Post #2679/ https://www.tonymacx86.com/threads/audio-hdmi-audio-applehda-guide.143760/page-268

malhal, 2017-08-31:
GA-H270 Gaming 3
[Malcolm’s iMac.acpi.zip](https://github.com/toleda/audio_CloverHDMI/files/1268679/Malcolm.s.iMac.acpi.zip)

toleda, 2017-09-01:
Edit dsdt is noted in the link, compile, save.  Install in EFI/CLOVER/ACPI/patched/
Restart, reply with new files .