toleda/audio_CloverHDMI #20: HD630 HDMI audio not working on Z270

DRenan, 2017-10-27:
Hi there, I've tried every posible way (known to me) to make the HDMI audio of my board work following the guide provided with the https://github.com/toleda/audio_CloverHDMI/blob/master/audio_cloverHDMI-130.command.zip

After reboot from installing 130.command I have no audio, nor devices.
Install CloverALC.command, reboot, audio but no HDMI ¿?

I've read a similar post where a user claimed he managed to make it work by removing HDEF dsdt patch, running HDMI patch, restart etc etc but I've not managed to make it work either.

I've seen people talking about a HDAS-2.aml or even -3.aml but I've not managed to make any of those.. 

Any ideas?
[ioreg+plist.zip](https://github.com/toleda/audio_CloverHDMI/files/1421758/ioreg.plist.zip)

toleda, 2017-10-27:
HDAS-1 only works when HDAS is present. When HDAS is removed by CLOVER/config.plist/ACPI/DSDT/patches/Item 0 (change HDAS to HDEF), no ssdt will work.  Remove the DSDT patch and restart. Clover will load the HDAS-1 ssdt as before and the installed ssdt will now work.

DRenan, 2017-10-27:
Hi there, thanks for the reply.

I'm afraid same output as before
[Z270-2.zip](https://github.com/toleda/audio_CloverHDMI/files/1422542/Z270-2.zip)

toleda, 2017-10-27:
Same problem. IGPU injection only works when GFX0 is present. When GFX0 is removed by CLOVER/config.plist/ACPI/DSDT/patches/Item 0 (change GFX0 to IGPU), ssdt will not work. Remove the DSDT patch and restart. Clover will load the HD630 ssdt as before and the expected injection will work.

DRenan, 2017-10-27:
Still no luck, I've even made everything from scratch (deleted all audio related kext/DSDT patches/KexToPatch/SSDT's from ACPI) and re run everything, just to make sure there was nothing blocking.. I Even removed a SSDT I have for the RX580 where the 'device' is GFX0 (since you mentioned the GFX0 thing)

Could it be a kext on EFI?
[Z270-3.zip](https://github.com/toleda/audio_CloverHDMI/files/1422821/Z270-3.zip)

toleda, 2017-10-27:
Motherboard?
Both ssdts are working
Attach cloverHDMI terminal output

toleda, 2017-10-27:
System kexts load from only one place, S/L/E.
The script patched the correct connector.
No audio is caused by no video.  No video is not an audio problem.

DRenan, 2017-10-27:
Sorry, I made a mistake, I meant 'AppleALC', which I have placed in EFI/Clover/Kext/10.3 restarted and now audio/video works through HDMI

There must be some reason as why video playback wont work if 'AppleALC' is not present in EFI folder.
And as why I had to run things twice¿? to make it work..

Can I help you somehow figure out if you want?

toleda, 2017-10-27:
AppleALC has nothing to do with video, an infeasible conclusion.

HD630 HDMI audio does not need/use AppleALC.kext, AppleHDA.kext is used for all HDMI audio. config.plist shows cloverALC ran; onboard audio likely worked before you installed AppleALC.  To be clear, the 887/layout 1 files are the same in cloverALC and AppleALC. 

Valid choices are:
1. cloverALC + cloverHDMI
2. AppleALC + cloverHDMI

Summary:
cloverALC + cloverHDMI + AppleALC = audio failure
Above has no effect on video.

DRenan, 2017-10-30:
Hi there, I know it doesn't make much sense, but it honestly doesn't work without the mentioned above..

If I run cloverALC on it own on clean install, after reboot I can see the devices, but no audio, and if I play a youtube video for instance, I get like a glitch where no audio plays, but no video plays either (video seems to load as I can see on load bar). As soon as I install cloverALC.kext in clover/kext/10.13 and restart, audio and video plays normally..

I know it doesn't make much sense but I'm real honest, it works like this.. At least on my build..

toleda, 2017-10-30:
macOS (10.12/10.13) does not support Kaby Lake/200 Series audio. AppleALC has a fix, cloverALC needs: https://www.tonymacx86.com/threads/quick-fix-200-series-kaby-lake-audio-onboard-and-or-hd-6x0-hdmi.221618/ There is nothing special, or peculiar, about your system. However, your "solution" is only supported by various incorrect assumptions.  When macOS supports Kaby Lake, likely with 10.13.1, expect audio to fail again.