toleda/audio_CloverHDMI #21: Device ID 0x16590000 not supported

jneira95, 2017-10-30:
Hi there Toleda, I'm sure you do not support all Devs ID on the market (that'd be crazy). I'm recently embarking on making a small hack for my nephew and have come across the message 'Device ID 0x16590000 not supported' when running cloverHDMI script..

Is there anything I could do to make it work?

The hardware in case is a MSI Cubi 2 (a NUK), which without HDMI audio is kind of disappointing lol.

Thank you so much

toleda, 2017-10-30:
For mobile processors, I rely on Users to help with the correct script settings. For 0x16590000, attach an IOReg with a HDMI device connected and the apci tableset (MaciASL/File/Export Tableset ...)

jneira95, 2017-10-30:
Sure thing mister! thx
[Archive.zip](https://github.com/toleda/audio_CloverHDMI/files/1428421/Archive.zip)

toleda, 2017-10-30:
IOReg shows no HD630 codec, i.e., no HD630 audio
Verify BIOS/ ... /Integrated Graphics/HD Audio/Enabled
Reply with new IOReg and config.plist

jneira95, 2017-10-31:
Hi there again, I've checked bios (very limited BTW) and HD audio is enabled.. I've used your cloverALC and I can confirm audio coming out of headphone jack.

toleda, 2017-10-31:
Your system has 2 codecs, one for headphone and one for HD630. I can confirm no HDMI audio possible until the HD630 audio codec is enabled.  Second request, find the BIOS setting and enable. Reply with new IOReg and config.plist.

toleda, 2017-10-31:
That is for onboard audio; Integrated Peripherals is the wrong place. Try System Agent Configuration.

jneira95, 2017-10-31:
I'm afraid there is no System agent... this Bios is a bit uhm, s**t..

toleda, 2017-10-31:
There is a section for CPU graphics, find it!

jneira95, 2017-11-02:
I'm afraid there isn't one.. Like I said, the bios is rubbish.. Perhaps a modified bios would unlock certain features?? should I look into that?

toleda, 2017-11-02:
Contact MSI Technical Support. User should have access to HDMI audio codec, system memory allocation, Render Standby, Multi-Monitor, etc.