toleda/audio_CloverHDMI #23: Intel HD615: Device ID: 0x1e590000 not supported

DottoreTozzi, 2018-11-06:
Hi,

I'm desperately trying to get HDMI Audio to work on the new LattePanda Alpha using Mojave.
Almost everything else works like a charm...
Here's my ioreg and acpi output:
[Archive.zip](https://github.com/toleda/audio_CloverHDMI/files/2553653/Archive.zip)

Cheers,
Stephan

toleda, 2018-11-06:
ioreg.txt is not helpful. IORegistryExplorer/File/Save a Copy As ...
https://github.com/toleda/audio_ALCInjection/blob/master/IORegistryExplorer_v2.1.zip

HD615 is not supported at the moment.  IOReg will determine if it can be added.

DottoreTozzi, 2018-11-06:
Sorry for that.
I hope this will help:
[ioservice.zip](https://github.com/toleda/audio_CloverHDMI/files/2555374/ioservice.zip)
Thanks for looking into this!

toleda, 2018-11-07:
What motherboard or device?
Do you have an emergency USB booter?
If yes, edit config.plist/Graphics/ig-platform-id/0x591b0000
Restart, reply with new IOReg, config.plist and Clover boot log.

DottoreTozzi, 2018-11-07:
Hardware is the brand new "Lattepanda Alpha 864":
https://www.dfrobot.com/product-1729.html
CPU: Intel m3-7y30 - Gfx: Intel HD 615 - Audio Codec: ALC269VC
My setup according to this:
https://youtu.be/nrJpwPxoZZ8

Emergency boot is no problem.
Here's the results:
[Archive.zip](https://github.com/toleda/audio_CloverHDMI/files/2556272/Archive.zip)

Many thanks again!

toleda, 2018-11-07:
Two major issues
a. No HD 515 HDMI codec. If available, BIOS/.../System Agent Configuration/Integrated Graphics Audio/Enabled (perhaps, CPU Display Audio)
b. New IOReg shows the correct HDMI connector, however, no connected display is reflected in IOReg, why?

DottoreTozzi, 2018-11-08:
I checked with Ubuntu Linux, the audio codec reported by /proc/asound/card0/codec#0 is:
ALC269VC.

The display wasn't connected directly, but through a Yamaha receiver. Both were switched on and proper input selected. Desktop showing. The setup works with Windows 10 and Ubuntu...

toleda, 2018-11-08:
Need an IOReg that shows the HD 615 HDMI codec and a display (Yamaha) connected.
Attach DPCIManager/PCI List

DottoreTozzi, 2018-11-08:
Sure thing! Should I switch to /ig-platform-id/0x591b0000 again before trying this?
Will connect HDMI directly to the LG TV this time, without the Yamaha in between.

toleda, 2018-11-08:
No. OK.

DottoreTozzi, 2018-11-09:
[Archive.zip](https://github.com/toleda/audio_CloverHDMI/files/2564094/Archive.zip)
Includes new ioreg, and screenshots from DPCIManager (didn't find any other export options there...).
This time, LG TV is connected directly. Unless the "pink screen fix" is interfering, which is basically a display override...(?)

toleda, 2018-11-09:
HD 615 HDMI audio codec now showing
Why was ig-platform-id/0x591b0000 replaced with IOReg/0x191b0000?
Try again, reply with new IOReg and config.plist

DottoreTozzi, 2018-11-09:
I used the latest "testing" build from novaspirit in this last one.
(Older one did no longer work with 10.14.1).
I saw only now that he also changed the platform id for whatever reason.
Guess he might be trying to figure out the same thing as we are right now, but isn't really in the mood to talk about it...

[Archive 2.zip](https://github.com/toleda/audio_CloverHDMI/files/2564222/Archive.2.zip)

Ioreg and current config.plist.

toleda, 2018-11-09:
novaspirit??
Same problem.
Reply with new files only with ig-platform-id/0x591b0000

DottoreTozzi, 2018-11-10:
Novaspirit is the guy who first came up with a working clover based solution for this particular machine. He has a config set he maintains on:
https://github.com/novaspirit/macpanda
Which is what my efforts are based on so far.

OK, I switched back to 0x591b0001 (which breaks gfx acceleration), connected the TV directly and this is the output:

[Archive.zip](https://github.com/toleda/audio_CloverHDMI/files/2567927/Archive.zip)

toleda, 2018-11-10:
We are not communicating
Never suggested 0x591b0001, not valid
3rd request, ig-platform-id/0x591b0000
2nd request, reply with new files only with ig-platform-id/0x591b0000

DottoreTozzi, 2018-11-10:
I'm really sorry, please stay patient with me.
I've been in IT for 30 years and have since shuffled some bits and bytes around, but regardless, Hackintosh is still almost completely new to me.
The (mis)leading (or tailing) 01 had been sort of a constant so far...

OK so I sincerely hope these will now be more helpful:
[Archive.zip](https://github.com/toleda/audio_CloverHDMI/files/2568054/Archive.zip)

BTW:
0x591b0000 breaks HDMI video output (no signal) and resets VNC screen sharing to 4:3 mode.

toleda, 2018-11-10:
OK, set ig-platform-id/0x591e0001. Reply with IOReg and config.plist

DottoreTozzi, 2018-11-11:
OK.
[Archive.zip](https://github.com/toleda/audio_CloverHDMI/files/2569380/Archive.zip)

toleda, 2018-11-11:
Add the 3 con1 patches to config.plist/Devices/Properties. Restart, reply with new IOReg and config.plist.  
More info: https://www.tonymacx86.com/threads/guide-intel-framebuffer-patching-using-whatevergreen.256490/
[image]

toleda, 2018-11-12:
HD 615 HDMI audio codec gone.
Why did you remove FakePCIID-Intel-HDMI-Audio.kext?
Install both FakePCIID kexts, restart, reply with new IOReg and config.plist.

DottoreTozzi, 2018-11-12:
It works now! Thank you so much for your patience!

[Archive.zip](https://github.com/toleda/audio_CloverHDMI/files/2573433/Archive.zip)